Alexander-T-Moss/Small-Area-Flow-Comp #2: Issue in Orca Slicer not exporting the changes

Needypoo, 2023-09-01:
I slice the file after adding the path to SmallAreaFlowComp_Win.exe "E:\OrcaSlicer-main\SmallAreaFlowComp_win.exe;"  to post processing scripts. I go to export it and a terminal pops up and then disappears. I then go to the gcode and search "old value" but it is not found.

neeamradia, 2023-09-01:
Please include platform and slicer version. Didn’t download the script from the repo or is it the discord channel one?

Needypoo, 2023-09-01:
I got it from both and they both resulted in the same.

[image]

[image]

neeamradia, 2023-09-01:
Do you have any other slicer installs (SuSi? PS?) that you can test with to make sure it’s not an issue with your PC or the script ? Could be a slicer issue tho as I don’t think we tested fully with Orca

Needypoo, 2023-09-02:
it does work in super slicer

neeamradia, 2023-09-02:
Thank you. Could you send a screenshot of Orca where you have entered the path please?

Needypoo, 2023-09-02:
[image]

neeamradia, 2023-09-02:
Ok. This is definitely an orca specific issue. Might be to do with the way it labels extrusion roles etc or the way it handles post processing scripts or even the way it outputs gcode to post processing scripts. Does klipper estimator work on Orca? After this bit of investigation, I think it’s in Alex’s hands to find the problem / solve it as I don’t have much more ability here.

Needypoo, 2023-09-02:
I am not sure but something funny i notice with Orca is it doesnt show how many layers are in the file.

neeamradia, 2023-09-02:
Thank you. Could be an issue. I'll leave this to @Alexander-T-Moss now as this has reached the limit of my knowledge and ability.

Alexander-T-Moss, 2023-09-02:
This should be something fairly easy to find, once I get the next revision out, debugging this will be a lot easier

Alexander-T-Moss, 2023-09-02:
This issue should be resolved in [V0.5.1](https://github.com/Alexander-T-Moss/Small-Area-Flow-Comp/releases/tag/v0.5.1)

neeamradia, 2023-09-03:
@Needypoo can you please confirm that this now works for you on OrcaSlicer?

Needypoo, 2023-09-07:
[image]
now with the latest one when trying to export gcode

Needypoo, 2023-09-07:
nevermind it must have been the model.txt file. working!

Alexander-T-Moss, 2023-09-07:
Okay, assuming this is working now I’ll close this issue :)

closed 2023-09-07 by Alexander-T-Moss