Alexander-T-Moss/Small-Area-Flow-Comp #7: OSX Arm 64 Artifact Runtime Crash

martinw89, 2023-11-05:
### Environment
- M1 MacBook Air 2020 on Sonoma 14.0 (23A344)
### Steps to reproduce:
1. Navigate to release: https://github.com/Alexander-T-Moss/Small-Area-Flow-Comp/releases/tag/v0.7.1
2. Download https://github.com/Alexander-T-Moss/Small-Area-Flow-Comp/releases/download/v0.7.1/SmallAreaFlowComp-v0.7.1-osx-arm64.zip
3. Extract files and navigate to extracted folder
4. Remove the Apple security nanny flag by running `xattr -dr com.apple.quarantine` on all files in extracted folder (SmallAreaFlowComp and SmallAreaFlowComp.pdb)
5. Run SmallAreaFlowComp executable. `./SmallAreaFlowComp <my gcode file>`
### Expected result
Script execution, modified gcode output
### Actual result
- A separate process kills the SmallAreaFlowComp process.
- zsh reports `zsh: killed     ./SmallAreaFlowComp`. (This does not indicate zsh killed the process, rather, it's in a format that means "zsh is reporting this process was sent SIGKILL".)
- `echo $?` returns 137.
### Logs
Unfortunately Console is not showing any crash reports for SmallAreaFlowComp. However, these 4 system log lines happen any time I try to run the executable:
```
680	error	19:06:31.645719-0500	syspolicyd	cannot open file at line 46973 of [554764a6e7]
680	error	19:06:31.645767-0500	syspolicyd	os_unix.c:46973: (2) open(/private/var/db/DetachedSignatures) - No such file or directory
680	error	19:06:31.646914-0500	syspolicyd	Unable to initialize qtn_proc: 3
680	error	19:06:31.647110-0500	syspolicyd	failed to call driver: 0x3
```
### Workaround
Cloning source and compiling creates a runnable executable.

martinw89, 2023-11-05:
Unfortunately I do development 100% on Linux in my day job so getting useful Mac logs is new to me. I already had to fight the Apply nanny to get those four lines in the system logs to not be redacted. If there are better logs I can provide that I'm just ignorant about, please let me know what to do and I'll be happy to report back!

chrisrichards, 2023-12-21:
I think this is due to code signing, can you try running the script below and then try running it?

`codesign --force --deep -s - ./SmallAreaFlowComp`

scriptnugget, 2024-06-08:
@chrisrichards I was having the same problem as @martinw89 and your command fixed it! Thank you! SuperSlicer successfully exported the gcode and passed it through the script. I checked the gcode with the analyzer at [gcode.ws](https://gcode.ws), and it seems to be working as intended.
[image]

Alexander-T-Moss, 2024-06-11:
Made a note to update the repo with troubleshooting having to sign the code for it to work on MacOS, will close this issue once that update is pushed