Alexander-T-Moss/Small-Area-Flow-Comp #8: Super high flow portions being added

teachingtechYT, 2023-11-29:
I'm having an issue with the script adding short very high flow segments, and possibly lowering the majority of the flowrate uniformly. Windows 10, Strawberry Perl installed as well as the script. Tested in SuperSlicer and OrcaSlicer. Both slicers add to the log without error and the gcode is modified. Everything appears to be running properly.
However, there is a problem with the modified gcode.
Before:
[image]
After:
[image]
Zoomed in:
[image]
This gives a predictable error from my firmware:
[image]

I'm not sure if that range of value skews the colour so much that the dark blue majority is an accurate depiction, but seemingly the majority of the print is now printing at ~4mm^3/sec.

Attached are the STL, plus the before and after gcode.
[gcode and stl.zip](https://github.com/Alexander-T-Moss/Small-Area-Flow-Comp/files/13505884/gcode.and.stl.zip)

Hopefully I'm just doing something wrong.

Alexander-T-Moss, 2023-11-29:
I believe I had a similar issue during initial testing, so will look into what caused this and see if I didn't fix it properly then

Alexander-T-Moss, 2023-11-29:
Can you try slicing with relative extrusion?

teachingtechYT, 2023-11-29:
Thanks for the quick reply.
Using relative extrusion fixes the bug, although the effect is very subtle:
[image]

Are there any variables the user can tune to change the extent of the compensation?

teachingtechYT, 2023-11-30:
Ok, so run the print as in rather than rely on the gcode preview?

CarnageMarkus, 2024-01-14:
I am also getting "Move exceeds maximum extrusion (2.104mm^2 vs 0.640mm^2)" while I enabled relative extrusion in superslicer. I run ender 3 on klipper. I too get random blobs of very high extrusion. Seems like there should be some kind of guard/sanity check that adjusted flows don't go into unreasonable values

Alexander-T-Moss, 2024-01-14:
Can you share the gcode :)