Alexander-T-Moss/Small-Area-Flow-Comp #9: Prusha slicer post prossing script doesn't exist

Kern-Knight, 2024-06-08:
[image]
[image]
[image]
Not sure what ive done wrong

Alexander-T-Moss, 2024-06-08:
You might need to put quotation marks around the path name as it contains a space, or remove the space by putting the file in a place where the path won't contain a space

Kern-Knight, 2024-06-09:
Ok cool ill try that
And get back to you
I did notice the space and was un sure as i only just learnt of this from a
youtuber

Kern-Knight, 2024-06-09:
I did the location in "quotations" and it worked a tread thank you for your
help and time today appreciate it and love your work making 3d printing
better and more reliable

Alexander-T-Moss, 2024-06-11:
Brilliant, glad that got it working for you, assuming all your issues are resolved I'll close this issue now, but feel free to re-open it or make another issues ticket if you need help with anything else, or something seems broken!

closed 2024-06-11 by Alexander-T-Moss