Alexander-T-Moss/Small-Area-Flow-Comp #16: Low flow rate for curved lines

Wupb, 2025-01-21:
Thin solid infill can be curved when concentric. In my case, it's circular when printing a ring shape. The reduced flow makes the wall not touch. I don't have pictures of the print, but can confirm the issue went away when turning off the compensation. Using Orca Slicer 2.2.0 and the default model.

###### SAFC on
[image]

###### SAFC off
[image]

Alexander-T-Moss, 2025-01-21:
This is interesting and something I haven’t thought off, so the way the slicer is drawing the curves here is in small individual segments, and due to the nature of SAFC, it’s seeing very short infill segments and thus is reducing the flow for each segment very aggressively, I’d recommend using a different type of infill and I’ll make a note on the repo about this as unfortunately due to the nature of how SAFC works I can’t see a way to simply solve this

Alexander-T-Moss, 2025-01-21:
You may also just need to turn it off as this model seems like an interesting edge case to cause this as normally this would be sliced as either having gap fill in it or just perimeters